3drobotics/ardupilot-solo #89: Solo is "Open for development", but the JIRA instance is closed to the public.

MAVProxyUser, 2015-06-12:
Solo is "Open for development", but the JIRA instance is closed to the public. 

http://diydrones.com/profiles/blogs/solo-open-for-development

End users and potential developers unable to see defect tracking such as: https://3drsolo.atlassian.net/browse/FC-186

proficnc, 2015-06-12:
we have met our obligations and all code is open.  There is NO obligation for 3DR to open any more than we have.

3DR use Jira for our internal communications, they are not public, nor should they be.

closed 2015-06-12 by proficnc

MAVProxyUser, 2015-06-12:
All code is not open... 
http://www.gnu.org/licenses/gpl-faq.en.html#RedistributedBinariesGetSource

The Solo App downloads binary firmware files that you have not released the source code to for example. Per GPL I have the right to ask for the source code to the binaries on the stm32 and the imx6 correct? or anything else contained on the squashfs for that matter? Is ArduCopter still GPL? 

Where is the source code for the following binaries? 
imx6solo-3dr-1080p.dtb
imx6solo-3dr-artoo.dtb

You've JUST now complied with releasing source to: firmware//ArduCopter-1.0.1.px4 which I applaud. 

What about all these? 

./usr/bin/app_connected_msg.py
./usr/bin/app_packet.py
./usr/bin/app_server.py
./usr/bin/app_streamer.py
./usr/bin/btn_client.py
./usr/bin/btn_msg.py
./usr/bin/buttonManager.py
./usr/bin/cable_cam.py
./usr/bin/camera.py
./usr/bin/checkArtooAndUpdate.py
./usr/bin/checkPixBaud.py
./usr/bin/checkUnlock.py
./usr/bin/clock.py
./usr/bin/configfile.py
./usr/bin/getmaclocal.py
./usr/bin/gimbal/firmware_helper.py
./usr/bin/gimbal/firmware_loader.py
./usr/bin/gimbal/firmware_release.py
./usr/bin/gimbal/firmware_to_header.py
./usr/bin/gimbal/firmware_version_header.py
./usr/bin/gimbal/setup.py
./usr/bin/gimbal/setup_comutation.py
./usr/bin/gimbal/setup_home.py
./usr/bin/gimbal/setup_mavlink.py
./usr/bin/gimbal/setup_param.py
./usr/bin/gimbal/setup_read_sw_version.py
./usr/bin/gimbal/setup_run.py
./usr/bin/gpio.py
./usr/bin/hostapd_ctrl.py
./usr/bin/hostapdconfig.py
./usr/bin/ifconfig.py
./usr/bin/infiniCable.py
./usr/bin/input_report_client.py
./usr/bin/input_report_msg.py
./usr/bin/ip.py
./usr/bin/ip_util.py
./usr/bin/iw.py
./usr/bin/led.py
./usr/bin/loadLog.py
./usr/bin/loadPixhawk.py
./usr/bin/location_helpers.py
./usr/bin/lockout_msg.py
./usr/bin/lsproc.py
./usr/bin/lsusb.py
./usr/bin/magfit.py
./usr/bin/magfit_delta.py
./usr/bin/magfit_gps.py
./usr/bin/magfit_motors.py
./usr/bin/main.py
./usr/bin/mavextract.py
./usr/bin/mavflightmodes.py
./usr/bin/mavflighttime.py
./usr/bin/mavflightview.py
./usr/bin/mavgen.py
./usr/bin/mavgpslock.py
./usr/bin/mavgraph.py
./usr/bin/mavkml.py
./usr/bin/mavlogdump.py
./usr/bin/mavloss.py
./usr/bin/mavmission.py
./usr/bin/mavparmdiff.py
./usr/bin/mavparms.py
./usr/bin/mavplayback.py
./usr/bin/mavproxy.py
./usr/bin/mavsearch.py
./usr/bin/mavsigloss.py
./usr/bin/mavsummarize.py
./usr/bin/mavtogpx.py
./usr/bin/mavtomfile.py
./usr/bin/miniterm.py
./usr/bin/modes.py
./usr/bin/mp_slipmap.py
./usr/bin/mp_tile.py
./usr/bin/MPU6KSearch.py
./usr/bin/orbit.py
./usr/bin/pair.py
./usr/bin/pair_button.py
./usr/bin/pair_confirm.py
./usr/bin/pair_server.py
./usr/bin/pair_solo.py
./usr/bin/pathHandler.py
./usr/bin/pixhawk.py
./usr/bin/px_uploader.py
./usr/bin/rc_cli.py
./usr/bin/rc_ipc.py
./usr/bin/rc_pkt.py
./usr/bin/rc_remap_sample.py
./usr/bin/RCRemapper.py
./usr/bin/roi.py
./usr/bin/rssi_send.py
./usr/bin/runlevel.py
./usr/bin/selfie.py
./usr/bin/shotLogger.py
./usr/bin/shotManager.py
./usr/bin/shotManagerConstants.py
./usr/bin/shots.py
./usr/bin/slip.py
./usr/bin/smtpd.py
./usr/bin/stick-axis-cfg.py
./usr/bin/stick-cal.py
./usr/bin/stm32_defs.py
./usr/bin/stm32loader.py
./usr/bin/telem_ctrl.py
./usr/bin/udhcpc.py
./usr/bin/updater_msg.py
./usr/bin/usb.py
./usr/bin/vector3.py
./usr/bin/wpa_cli.py
./usr/bin/wpa_control.py
./usr/bin/wpa_supplicant.py
./usr/bin/yawPitchOffsetter.py

Are the above python scripts considered not to be "Code that we collectively decide provides greater value to our community of developers than the incremental value it may provide to 3DR if kept proprietary"? 

http://3drobotics.com/open-source-policy/

Good job on recently updating http://3drobotics.com/open-source/ BTW.