3drobotics/ardupilot-solo #103: Drift mode loses altitude during travel when throttle is at 50%

dngrlux, 2015-06-19:
This may be expected behavior… please close if so.

With the Solo controller defaulting to hover at 50% stick/throttle, I had a possibly-false expectation that the altitude would not deviate too much when using drift mode. In today's tests I brought the Solo to 400' AGL and tested full forward flight and banks in drift mode, and the altitude dropped over a period of 10-15 seconds to 350', then to 300'. If the FLY mode can be tuned to hover at 50% throttle, could some kind of throttle curve or barometric assist be utilized to keep the quad in drift mode at generally the same altitude without throttle input?

With the IRIS+ in drift mode, I had no expectation of holding an altitude at precisely 50% throttle, as the stick is not spring loaded and manual throttle position is required.

nick3dr, 2015-06-21:
https://3drsolo.atlassian.net/browse/FC-145 <https://3drsolo.atlassian.net/browse/FC-145>

@dngrlux I don’t recognize your username. Who is this?

This issue is OK to be closed as it is tracked in Jira.

dngrlux, 2015-06-21:
Hi Nick, I'm just an end user posting an observation. Previous experience piloting micro quads, miniquad FPV racers, IRIS+, and now Solo. UI/UX designer by trade.

nick3dr, 2015-06-21:
Oh, great! Thanks for your input!

proficnc, 2015-06-21:
this should be left open, and linked, so users can see the final outcome here.  An issue should only be closed in Git when there is a publicly available closure to an issue.

brettdaman, 2016-09-20:
+1 love the drift mode but dont trust it because of the altitude gain or loss depending on the battery level.