3drobotics/ardupilot-solo #131: Delete closed branches

OXINARF, 2015-07-16:
Can't branches that have closed PRs be deleted here in GitHub? There's already 134 branches...
When a branch has a PR it can be deleted without any information being lost. Also if needed the branch can always be restored later.

rmackay9, 2015-07-17:
What might help is if the developers (primarily Jon and Sid) made private forks of this ardupilot-solo repo and did their development there and then raised PRs from these personal repos into ardupilot-solo/master.
That would be more consistent with how the main ardupilot branch is utilised but really, it's up to Paul, Sid, Jon's better judgement about how best to work.

OXINARF, 2015-07-17:
Randy that would be an option, the best option I think.

wsilva32, 2016-09-23:
Agreed that this would have been a better way to go. I'm going through and cleaning up this repo now, so I'll be cleaning up branches soon as well. :)

closed 2016-09-23 by wsilva32