3drobotics/ardupilot-solo #210: ardupilot-solo + PX4Firmware-solo not building

Joeyn1993, 2015-11-29:
Hi guys,

I'm currently trying to build the ArduCopter code for a 3DR Solo.

I've cloned 
https://github.com/3drobotics/ardupilot-solo/
https://github.com/3drobotics/PX4Firmware-solo
and github.com/diydrones/PX4NuttX (new users only allowed 2 links sorry)
into a directory and renamed PX4Firmware-solo to PX4Firmware.

I then run 'make px4-v2' or 'make px4' from the arudpilot-solo/ArduCopter directroy and the build proceeds for a while before getting the error:

/Users/Joey/GitHub/soloTest/ardupilot-solo/libraries/AP_NavEKF/AP_NavEKF.cpp: In member function 'void NavEKF::CovariancePrediction()':
/Users/Joey/GitHub/soloTest/ardupilot-solo/libraries/AP_NavEKF/AP_NavEKF.cpp:1901:1: error: the frame size of 2256 bytes is larger than 1300 bytes [-Werror=frame-larger-than=]
}
^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
make[3]: *** [libraries/AP_NavEKF/AP_NavEKF.cpp.o] Error 1
make[2]: *** [/Users/Joey/GitHub/soloTest/PX4Firmware/Build/px4fmu-v1_APM.build//Users/Joey/GitHub/soloTest/ardupilot-solo/module.pre.o] Error 2
make[1]: *** [/Users/Joey/GitHub/soloTest/PX4Firmware/Build/px4fmu-v1_APM.build/firmware.px4] Error 2
make: *** [px4-v1] Error 2

Is there anything wrong with my build process here? Or does anybody know a fix for this? Any help would be really appreciated!

alexblack, 2016-03-21:
Anyone figured this one out?

wsilva32, 2016-09-23:
@Joeyn1993 I know this is quite a late response, but we maintain our own fork of PX4NuttX located here: https://github.com/3drobotics/PX4NuttX.

If you're still having issues building, could you post some details about your build environment? e.g. operating system, g++ version, etc...

alexblack, 2016-10-05:
I solved this by building on Ubuntu instead of OSX

wsilva32, 2016-10-05:
@alexblack Ubuntu is usually the preferred way to go :) 

I have tested this on OS X (Sierra) and it builds OK with the proper gcc toolchain setup. Things change pretty quickly in that world because of XCode and XCode Dev Tools..