3drobotics/ardupilot-solo #212: Has any one run solo using PPM input?

limhyon, 2015-12-20:
Hi,
I found that current solo PPM input has some electrical hardware glitch.
When I connect PPM via solo main board, the signal is mixed (checked by oscillo scope)
I think it is due to the buffer, mixed with AUX OUT.
Can someone check this problem?

Gussy, 2015-12-20:
Can you post a scope trace of the input signal while it's not connected to Solo, and another of the "mixed" signal you are seeing?

A picture or diagram of how you're connecting to the Solo main board would be useful as well.

mtbsteve, 2015-12-20:
+1. 
A picture how you connected to PPM would be great. 
I was not aware that PPM in is available on the main board.

proficnc, 2015-12-20:
Some detail would be good, what rx are you using? Have you removed the iMX6? 
I fly with a ppm RX from FR Sky, and have no issues...

wsilva32, 2016-09-23:
Closed due to no response from OP.

closed 2016-09-23 by wsilva32