3drobotics/ardupilot-solo #213: Safety request for cablecam - Exit into fly and climb on full throttle

IanHudsonP13, 2015-12-30:
When setting up a cablecam you make assumptions at the time of recording your cable, for example you'll record A to B whilst it's clear of obstructions, such as members of the public, vehicles etc.

When executing a cablecam, it would be desirable to have full throttle as a means of immediately exiting out of cablecam into fly and climbing allowing a very fast reaction to an unexpected problem, say a member of the public or vehicle entering into the path of a cable.

The time it takes to look down at the controller and hit pause or fly is 1 to 2 seconds, it's a lot of time in a panic situation and goes against the idea of keeping your eyes on the drone at all times which commercial operators are instructed to do.  At least with orbit it reacts to throttle so you can gain height to avoid a situation, the pause and fly buttons simply aren't located all that well for a panic.

nick3dr, 2016-01-02:
Thanks for the suggestion. It's a good suggestion to solve a need that I can relate to. Unfortunately there's a challenge in that the left stick is remapped to camera tilt in the Cable Cam shot, and users wanting to tilt the camera up would not want to exit the shot.

We'll think about how to solve this problem, keeping all possible implementations in mind. I guess the problem at the core is that you want a quicker "panic" interface than pressing the FLY button?

IanHudsonP13, 2016-01-02:
Isn't tilt also on the left paddle so does it really need to be on the left stick as well (TBH I didn't even realise the left stick was being used for tilt)?

Yes, the core issue is needing a panic mode way out of a shot that doesn't involve looking down at the controller.

nick3dr, 2016-01-04:
Okay, thanks for the suggestion. We'll think about what to do here. Holding your finger over the FLY button might be your best bet for the next little while.

You should try out the left stick pan/tilt in Cable Cam. I love it! It's especially cool if you are tracking a moving subject and you set "Automatically point camera" to OFF in the Cable Cam settings.

closed 2016-01-04 by nick3dr