3drobotics/ardupilot-solo #215: Unresponsive after setting SERIAL1_PROTOCOL to "0"

rmackay9, 2016-01-15:
Even knowing that this was a bad idea I set SERIAL1_PROTOCOL to "0" on my solo.  Even after a factory reset I could not get the controller (or mission planner running on my desktop) to talk to the Solo.
I believe the only way to recover from this state is to disassemble the solo and connect to the USB port.

johnfinley, 2016-01-15:
If that is the telemetry port, and that setting is disabling hw flow control (are both of those things right?), then it *may* be the case that you could disable flow control in /etc/sololink.conf and it would work well enough to avoid disassembly.

peterbarker, 2016-01-15:
SERIAL4 also does mavlink.  I think that's the gimbal, and that's externally accessible.

proficnc, 2016-01-15:
Load plane firmware via ssh, it uses USB to do that... Then go back to copter..

closed 2016-09-23 by wsilva32

d13-kevinf, 2017-02-12:
SERIAL2_PROTOCOL anything other than zero will do the same thing FWIW...