3drobotics/ardupilot-solo #217: Get GPS data: Signal-to-noise ratio 3DR Solo

santiandres33, 2016-02-23:
Hello all,

I have acquire 3DR Solo drone, I havent figured it put how to obtain GPS data en NMEA format, specifically SNR and PNR values of each satellite? Is it possible to do this with the u-blox chip??

Thank for your time answering if possible.

Regards,

Santiago

hamishwillee, 2016-02-23:
This is not the right place to ask generic questions about Solo. Try the [discussion boards](https://discuss.dronekit.io/c/solo-development)

That said, AFAIK the autopilot does not expose the internal output format of GPS systems but instead shares whatever information it gets as MAVLink output. So you can probably get the information you need from http://mavlink.org/messages/common#GPS_RAW_INT and http://mavlink.org/messages/common#GPS_STATUS

I'm closing this issue as "invalid" for this location. Good luck with your development!

closed 2016-02-23 by hamishwillee

santiandres33, 2016-02-25:
Thank for the information, it is the most helpful I have found so far.