3drobotics/ardupilot-solo #219: Seemingly mismatch between ArduPilot's Dataflash format.

MadisonBlake, 2016-05-31:
Comparing 3DR's dataflash parser [here](https://github.com/3drobotics/ardupilot-solo/blob/master/libraries/DataFlash/LogStructure.h) and Ardupilots dataflash parser [here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogStructure.h), the two produce vastly different dataflash logs, specifically the GPS tag.

It would be nice if the two would create identical logs.

It seems the ardupilots format makes a lot more sense as "TimeUS" is always used to transmit internal clock time. While in 3DR's format "TimeMS" typically sends the internal clock, except in the GPS tag, where it sends the MS of the current GPS week while a separate tag, "T", sends current clock.

rmackay9, 2016-06-01:
@MadisonBlake, I think the difference is because solo master is based on a slightly older version of ardupilot.  For the most part ardupilot master runs on the Solo but there are still a few missing bits.  The ones that I know of are:
1. ESC sync failure recovery
2. ESC Leds

The 2nd one at least is in a PR here but it's out of date now:
https://github.com/ArduPilot/ardupilot/pulls?q=is%3Aopen+is%3Apr+label%3ASolo

wsilva32, 2016-09-23:
@MadisonBlake Yes, our version of APM was forked from AC 3.3-dev quite some time ago. While we have backported many of the changes to upstream, the reverse has not been as easy. If you're interested in flying latest APM master and its newer features on Solo I'd recommend checking out https://github.com/SoloSmartDeveloping/ardupilot where some folks from the FB user group have started developing a port.

closed 2016-09-23 by wsilva32