3drobotics/ardupilot-solo #274: lading with drift mode doesnt stop rotors.

brettdaman, 2016-09-20:
Love the drift mode when flying with a fixed camera.
Only if I try to land with drift mode I can keep the throttle stick low for seconds and the rotors do not turn off.
On landing with the throttle stick down, I needed to switch to fly mode and then they turned of instantly.

nick3dr, 2016-09-20:
Landing is only supported in some modes:

Stabilize, Alt Hold/ Fly:Manual, Loiter/Fly, Land, Return Home.

Please switch to one of these modes before landing.

closed 2016-09-20 by nick3dr

rmackay9, 2016-09-21:
@brettdaman, I've seen this in master as well.  I intend to fix it eventually but it won't make it into AC3.4.  How that fix could eventually end up on a Solo is a question mark at this point.  It is certainly possible to load master onto a Solo but it would likely need testing and some changes.
Here's the master issue in case you want to follow it: https://github.com/ArduPilot/ardupilot/issues/2300

brettdaman, 2016-09-21:
@rmackay9 thanks for taking a look at this issue, just as in ardupilot issue it happend while I gave the controller to a friend, myself I always land on fly, no crash but could turn very bad.
Do you now what the latest AC release that is merged in solo is?

If it is a "feature" like @nick3dr suggest it should be more clearly stated in the manual or in the description of the mode in my opinion.