3drobotics/ardupilot-solo #284: Not really a ardupilot-solo issue, but controller i guess..

AndKe, 2017-01-23:
Sorry for posting here, trying to get in touch with whoever still have some source, or the knowledge about Solo /Controller internals.

The problem:
Onboard Solo , it's possible to read all the controller buttons: 
like:
power.press
power.release

For more advanced payload, I tried to use doubleclick
it looks like:
power.press
power.release
power.double
power.release

- but once that happens, everything gets reversed:
a press is now "release" first, then "press".

That bug (wherever it is) makes it impossible to properly differentiate between press, and double.

That's not the whole story, after doubleclick, the doubleclick may be sent again on next single press. (several seconds later)  - but only first time.

I'd like to get rid of this bug, most likely in the controller, something after a doubleclick is not reset properly.

dcarpy, 2017-01-27:
Hi Andre.
I know this has nothing to do with your question/issue, but was curious if you are running 1.5.2 of the firmware on your Solo?
Dave

AndKe, 2017-01-27:
@dcarpy yes, it's 1.5.2 - I did a bit of testing, and bad weather stopped me before >1h of flights , but will do much more next week (weather should improve)  - will be testing an RX100v2 as payload,  but that's not affecting AC.  (it's stock 1.5.2, no customizing there.)  Are you using Solo ? - feel free to contact me by mail on gitter.im  :)