3drobotics/ardupilot-solo #287: ardupilot-solo firmwares

AeroHawk, 2017-02-15:
Hi,
Does somebody know the difference between ardupilot-solo and ardupilot-solo-rebase? I have trying to use the last one but the 3DR Solo does not respond as expected: in auto mission it seems that mission can not be loaded.
Thank you

mtbsteve, 2017-03-13:
The rebase code uses Arducopter 3.3 whereas the official current version is based on Arducopter 3.2.x
I am using the rebase code on one of my Solos and have not experienced any issues so far. 
Maybe you could provide more details about your problem, for me it works also fine with auto missions in Tower.

rmackay9, 2017-03-14:
It seems to be a common misunderstanding that Solo uses (or used) AC3.2.x as a base but the official Solo firmware is somewhere between AC3.3 and AC3.4.  It's never been as old as AC3.2.1.

mtbsteve, 2017-03-14:
Thanks for clarification Randy. Some of us including me got likely distracted by the introduction of all the nice 3.3.3 features like Lidarlite i2c support, new vibe dataflash messages,  and so on with the Solo rebase code last year.