3drobotics/ardupilot-solo #288: Solo Version

porlock, 2017-02-19:
This is not an issue in direct way, more like a dispel doubts,
Does this firmware repo suits only for industrial Site Scan Solo version or it is also for earlier GoPro "civil" version ?

kevinmehall, 2017-02-19:
Version 1.3.1 in this repo corresponds to Solo release 2.4.2, which is the latest that has been tested by 3DR on GoPro Solo hardware with the Solo mobile apps. More recent versions have only been tested and released with SiteScan.

porlock, 2017-02-20:
That is to bad, GoPro Solo is not an old product and already obsolete without official firmware updates. Any way I fully understand that recent versions are tested only on SiteScan version, but is there any main reason that it should not work correctly or make some damage on GoPro Solo ?