3drobotics/ardupilot-solo #290: Can not build Ardupilot-solo.

fenyi2016, 2017-04-05:
Hi there,

I followed the instructions here to build ardupilot/ArduCopter successfully on ubuntu, 
http://ardupilot.org/dev/docs/building-px4-for-linux-with-make.html#building-px4-for-linux-with-make 

but it seems not working with ardupilot-solo/ArduCopter and I got this error after I ran 'make px4-v2': 
../mk/environ.mk:120: *** ERROR: could not locate /c/Users/sxt/Desktop/ardupilot-ArduCopter-3.2/ardupilot-ArduCopter-3.2/config.mk, please run 'make configure' first and edit config.mk. _Stop.

Then I ran 'make configure' & 'make px4-v2' and I got a new error: 
../mk/px4_targets.mk:30: *** ERROR: NUTTX_SRC not set correctly - no configs directory found.  Stop.

Does anyone know about that? Any input will be appreciated.

Tenacity-soar, 2018-10-10:
Do you solved your problem?
I want to know how to build Ardupilot-solo.

peterbarker, 2018-10-10:
On Tue, 9 Oct 2018, Tenacity-soar wrote:

> Do you solved your problem?
> I want to know how to build Ardupilot-solo.

I recently built opensolo simply by bringing up the Vagrant VM and running 
the supplied build script.

Note the "clean" gotcha - when re-running the script, consider specifying 
the option such that it doesn't clean any work it has previously done!