3drobotics/ardupilot-solo #293: mRo GPS M8N

n01knows, 2017-05-03:
Hi Will,

I noticed that on last published by 3dr solo firmware 1.3.1 I got around 20 sats on mRo GPS because besides of GPS satelites it could see also Galileo sats.
When I upgraded to 1.5.3 I was able to see 16 sats in the same location.
I checked with my colleague and he also told me that on 1.3.1 can see more than 20 sats.

I own 2 solos with mRo GPS m8n - https://store.mrobotics.io/product-p/gps001-mr.htm and both solos are reporting 16 sats when previously they could see 20+ sats.

I've contacted Jordi from mRobotics and I pasted his reply below:

"Hello Tomasz, 

We will check it out but maybe is because firmware 1.5.3 is resetting the GPS and by resetting the GPS the Galileo constellation gets deactivated."

Can you pls restore functionality for Galileo sats? Signal from Galileos are stronger than from GPS.

mrpollo, 2017-05-11:
Hey @n01knows thanks for the report, we'll try to reproduce this and follow up with next steps

mirandafly, 2017-05-26:
I'm using the firmware 1.5.3 and used two GPS (Drotek GPS, Radiolink GPS).  Both can get 18~20 sats before takeoff and during the flight without any problem.    

[image]

[image]