3drobotics/ardupilot-solo #295: Smart battery

Frederic88, 2017-06-14:
Hi,

In this version and any version of the solo firmware, the SMART batteries data are not displayed in the LOG files (TLOG or dtaflash BIN files). These data are very important to monitor the batteries.

Would it be possible to have an update with these infos  ?
 Get update from the SMbus (data like Battery serial number, cell voltage, temperature... ) and send that to the LOG throught a mavlink message.

Thanks,
Frederic

rmackay9, 2017-06-15:
Just FYI - there's a project to "get solo back on master" which means running Solo on unmodified ArduPilot.  I'm not leading the project but it seems like AC3.5-rc8 which I'm pushing out today for beta testing does make it possible.  This includes logging of the individual cell voltages.  I'm afraid it's not a free lunch though because for the moment, we recommend upgrading the Solo's flight controller to the Green Cube (aka Pixhawk2.1 special edition for the Solo).

ArduPilot discussion: http://discuss.ardupilot.org/t/testing-3-5-on-my-solo-she-flies/15655/56
There are a few Solo facebook groups that also might talk about this project.

Anyway, definitely not trying to push you in any particular direction.  A new release of 3DR's ardupilot branch would be an easier solution for you.  Just wanted to highlight this option.

Frederic88, 2017-06-15:
Hi,

Thanks very much for your input. 
Yes I had a look to this project "Get solo back on master" look very promising :
http://discuss.ardupilot.org/t/update-on-mastering-solo-major-milestone-reached/17959
But as you said you need to install the green cube... 
I was looking more for something that would be more an  "official release" and backed by 3DR robotics. 

Thanks again
Frederic

mtbsteve, 2017-06-15:
Frederic,

I tested 3.5 RC8 on Solo today. Smart battery logging and reporting in Solex works very well. 
You may join the Solo beta testing group on fb for the latest news and hints

anthonysleck, 2017-06-15:
mtbsteve,
Was that with or without a green-cube?

jmachuca77, 2017-06-15:
Be aware that Solo is no longer being developed by 3DR no new features will be added, no new “official” releases will be made (unless there is something critical which I very much doubt). Your only chance to get this is to use master. But as you know there is a piece of code missing from master that protects the ESCs from killing themselves. 

Best regards,
 
Jaime Machuca <mailto:jaime@element.aero>
CEO | Element Aero | www.element.aero <http://element.aero/>

 <http://element.aero/>

proficnc, 2017-06-15:
Adding the green cube is just one way to fix the issue, 
You can add a servo level shifter to each esc, or swap out the ESCs.

The green cube however is known to work. The additional features gained by this, and the massive ongoing support in the community make it quite a compelling option. 

There is also work ongoing to make this functional with standard hardware. It will however be at reduced performance

jmachuca77, 2018-08-16:
This has already been implemented in ardupilot master.