3drobotics/ardupilot-solo #297: Rebase on latest arducopter 3.5+?

tribbloid, 2017-08-05:
With the new OpenSolo initiative & solo link open sourced. A rebase on latest branch can bring numerous features introduced after copter 3.4+ and make R&D work on it much easier (because we don't have to maintain compatibility with 2 versions). is there a roadmap and timeline for this?

peterbarker, 2017-08-05:
On Sat, 5 Aug 2017, Peng Cheng wrote:

> With the new OpenSolo initiative & solo link open sourced. A rebase on
> latest branch can bring numerous features introduced after copter 3.4+ and
> make R&D work on it much easier (because we don't have to maintain
> compatibility with 2 versions). is there a roadmap and timeline for this?

I can't speak for 3DR here, but I don't think 3DR will be spending 
resources on moving anything from ardupilot-solo to ardupilot's 
master branch.

Even if an effort was made - it won't be by rebase.  Several people made 
some titanic efforts to rebase (and validate) the rebase, unsuccessfully 
in the end.

Are there specific features from the ardupilot-solo tree you think would 
bnefit ArduPilot's master?  If you can enumeate and make the case for 
specific pieces, the Solo community might step up and attempt to pull them 
in.

mtbsteve, 2017-08-06:
@tribbloid just go for AC3.5 and you have everything in there from 3DR combined with all the new stuff of Arducopter. . 
All details here:
http://ardupilot.org/copter/docs/solo_arducopter_upgrade.html

tribbloid, 2017-08-06:
@mtbsteve thanks a lot!
Are all the additional features (e.g. light) works?
though I probably won't do it immediately, green box takes too much time to install.

tribbloid, 2017-08-06:
@peterbarker, you are right, probably won't worth it.

tribbloid, 2017-08-09:
@peterbarker BUUUUT we still have several projects in dronekit that can be detached from 3DR firmware, e.g. dronekit-sitl no longer have to download 3.3 any more.

peterbarker, 2017-08-10:
On Wed, 9 Aug 2017, Peng Cheng wrote:

> @peterbarker BUUUUT we still have several projects in dronekit that can be
> detached from 3DR firmware, e.g. dronekit-sitl no longer have to download
> 3.3 any more.

That's actually more a matter of someone building and testing more 
binaries for dronekit-sitl than anything else.

I typically use dronekit-sitl by pointing it at my own compiled binaries - 
which is what I usually want to test, after all!