3drobotics/ardupilot-solo #300: how to build ardupilot-solo

Tenacity-soar, 2018-09-30:
I have changed ardupilot-solo's code , I don't know how to build it.

j07rdi, 2018-10-02:
Hello @Tenacity-soar 

I was able to do some compilation before. Maybe you want to reach Open Solo chat room, they might be able to help. 

We have developed a spin-off the SOLO board and I was able to compile some stuff, but I'm still having some issues here and there. 

Warmly,

proficnc, 2018-10-02:
As Jordi suggested, maybe have a look at the Open Solo project, if you are on Facebook, there are many in the solo community that may be able to assist you, or maybe you can ask on the Ardupilot forums.

Tenacity-soar, 2018-10-08:
thank you very much. I don't have a facebook account. I have made some attempts. But i met some problems.  My goal is to generate the ArduCopter-v2.px4 file.
1.git clone https://github.com/ArduPilot/ardupilot
2.cd ardupilot
3.git submodule update --init --recursive
4.Tools/scripts/install-prereqs-ubuntu.sh -y
5.. ~/.profile
6../waf configure --board px4-v3
7../waf copter
then i get the arducopter.px4 file
I use the scp command to upload the arducopter.px4 to  root@10.1.1.10:/firmware/
then i reboot the 3DR solo UAV. But some problem happened.   
I guess the reason is that ardupilot codes is not suit for the 3DR solo UAV.
So i want to know how to build the ardupilot-solo and generate the ArduCopter-v2.px4(https://github.com/OpenSolo/ardupilot-solo/releases)

Can you help me ?  thanks